midilab/aciduino #1: FEATURE REQUEST (not an issue): Easy x0xheart integration with 8 note knobs and note display. Thanks in advance!

01GOD, 2019-08-28:
Hi, 

Would be great to make this easy to use with the x0xheart project as a full sequencer. Add a nice little blue note display and 8 step knobs for notes. 

Thanks in advance!

01GOD, 2019-08-28:
> The initial roadmap was about to put it on a eurorack too, so midi and cv
> was my initial intent. But im out of any cv/gate equipment at the moment,
> what makes it really hard to develop and test this new cv/gate feature to
> integrate with xOxheart for example.
>
> Maybe if i got some help from a tester with a xOxheart or any other acid
> equipment with CV/Gate that can test and debug it for me it will be an
> option for now! ;)
>
> I got a display in mind also for this feature(128x68 mini oled) and im
> planning to release a DIY PCB with support for that.
>
> The ideia of knobs for notes are really killer! my initial plan was to
> make it like you request, but the design and constraints of the
> project(specially avaliable ports and space on protoboard) have guide the
> project into a more compact interface.
>
> I have some PCB ideias already to make it happen(it will be aciduino v2),
> but for protoboard version of aciduino v1 there is no way to add more
> buttons, knobs or display unfortunally because of the lack of avaliable
> ports and space on protoboard.
>
> For PCB version i will handle a display and more knobs/buttons by using
> multiplexers to extend it.

Oh I see. Here are some things found during research. (I don't have a
x0xheart handy yet)

Have a look at the "punkseq10" here. There is a build video.
https://www.whitefordresearch.com/products/punkseq10-preorder?variant=45657478407

Also look at this with a little display:
https://www.youtube.com/watch?v=emMyMncFpMk

Keeping it SIMPLE on the electronics build side is DEFINITELY recommended.
That is one thing that is really cool about this. Modding things in
software logic is much easier to handle later.

Adding a "7 segment display" or LCD (because OLED has ghosting issues) in
the center of that punkseq10 and adding slide and accent would be a nice
way forward.

Looking forward to trying an aciduino 2. Thanks in advance for making it an
awesome sequencer! Honestly the community really needs a proper DIY
sequencer for the x0xheart and MB33 Retro.

midilab, 2019-08-29:
This little display on the video is a OLED one, the same i was told you, but with less pixel space(128x32 vs 128x68). I had manage to work with that display without ghosting issues(most of the times it happens because of the shared interface of SPI or I2C with another subsystem), it works really nice and has a good visibility for night and day.

As soon as i have time to, i will try to finish the PCB for the aciduino v2, some changes like swap the arduino nano for a ardunio pro micro to have a USB/MIDI connection that will help people integrate computer setup with standalone acid basses, or even virtual acid basses on PC will be an easier setup option.

This project is part of a bigger project(not released yet) for a complete full multitrack DIY sequencer using arduino mega(16 tracks, 4 midi in/out, usb/midi, sdcard support for pattern save/load, pattern matrix remix - like ableton scene mode - and other goodies for sequencer dudes out there), just like a midibox seqv4 one.

My bad i had miss a MB33 deal last year... :( hope to get myself a xOxheart soon!

SeanTurner026, 2020-04-13:
> This project is part of a bigger project(not released yet) for a complete full multitrack DIY sequencer using arduino mega(16 tracks, 4 midi in/out, usb/midi, sdcard support for pattern save/load, [...]

Has this one come out yet? Otherwise, aciduino is a cool project! I've wanted to do something like this for a few years now

doctea, 2020-04-13:
Chiming in too because this is also one I have been hopefully waiting for? :)

midilab, 2020-04-13:
> Has this one come out yet? Otherwise, aciduino is a cool project! I've wanted to do something like this for a few years now

The 8 tacks sequencer is not released yet, but it is sitting here with me, my personal sequencer for loop music nowdays.

[image]

midilab, 2020-04-13:
> Chiming in too because this (that is, the 'aciduino v2 pcb') is also one I have been hopefully waiting for? :)

Im just trying to get some time to continue my work on it, but for sure it will be released this year...
For now some drafts i came up so far for the PCB thing, getting smaller and easier to make, by using a OLED screen we can minimize the interface for a clean design(the CV Converters goes on the middle of protoboard)
I had pickup the arudino pro micro because of his in-built capatibilities to act as a standalone midicontroller also for USB on computer side(avoid the use of serial to midi converters on the way). So that said, we can easily integrate our CV modular setup with computers for clock sync and other goodies.

[image]

midilab, 2020-05-23:
Finished the final layout of aciduino V2.

On the empty middle we can chose go for:

* CV/GATE 2 channels 12bits  
* MIDI Input  

Still working on CV/Gate module...

[image]

01GOD, 2020-05-24:
Saw now. Nice! Thanks!

If I can get a x0xheart for a decent price, likely make one of these soon.

midilab, 2022-10-05:
First steps almost done.

Schematic is ready and the source code almost done.

The idea behind the minimal is to make it esiear to build but also to extend, here is where it goes the cv/gate request for modular that will be coded soon.

[https://github.com/midilab/aciduino/blob/aciduino_v2/AciduinoV2/aciduino_v2.png](aciduino v2 schematics)

midilab, 2022-10-14:
closing this issue since the release of aciduino 2 is comming.

first steps is the basic module, then you can extend it for more detent encoders or pots as you wish

closed 2022-10-14 by midilab