midilab/aciduino #2: How do potentionmeters connect exactly?

SeanTurner026, 2020-08-30:
Hi,

Really think this is a cool project, but having difficulty finishing putting it together specifically with regards to the potentionmeters. 

Can someone give me locations on the board where the connecting pieces go? E.g. `J3`.

The potentionmeters in the picture on the repo don't indicate where the connectors go.

The screenshot below from the video doesn't indicate where to connect exactly.
[image]

Any help with this would be amazing. Thanks.

midilab, 2020-08-31:
welcome aboard!

the fritzing schematic picture on the repo has the pinage, by taking a looking now its not so visual the way it compiles the final image version, so you're bring here a point.

the connections that fritzing software had created are over the potentiometer component, but you can see a pattern, blue(ground), gray(signal) and red(5v) from the left to rigth side on repo picture, and they are connected to the 3 pins of pontentiometers. 

the potentiometer on the video image are different from the repo picture only by pins distance(those are easly found where i live, but the one on repo are most common avaliable across the globe).

but you can use the fritzing file on repository to check the schematic

midilab, 2020-10-20:
closing because of inactivity...

closed 2020-10-20 by midilab