midilab/aciduino #3: 303duino+aciduino = TD-303

Harvie, 2021-01-25:
Aciduno has 303 sequencer, 303duino has 303 synth. Why not combine these two to get full 303 experience?

https://github.com/midilab/aciduino
https://github.com/treisti/303duino

midilab, 2021-01-26:
looks like a great idea.

303duino makes use of mozzi library for audio sound. i left uClock(the bpm generator of aciduino) a commented code to make use of other timers(since mozzi makes use of timer1, the same as uclock), but never test along with mozzi library, but thats on my todo list.

I was asking myself how it sounds like, mozzi is a really good library, but how 303ish it can be?

maybe if a get some time to test the 303duino i can integrate booth.

Did you test it? can you make a audio or video so i can check how it sounds?

Harvie, 2021-01-26:
I have plans to test it, but i guess the sound mostly depends on output method you choose. Mozzi supports everything from pwm to external i2s dac.

midilab, 2021-01-26:
true, im working rigth now on aciduino v2, using teensy LC, that has a DAC builtin and mozzi support out-of-the-box. that can be a good choice for a cheap ARDUINO-303.

LeandroBarros, 2021-04-02:
I was thinking if it's possible to get MIDI clock input? That would make this even more useful to integrate with other gear in the studio.

midilab, 2021-04-03:
@LeandroBarros yes its possible, i didn't make aciduino v1 midi input ready to keep it low cost and simple to construct as possible(since midi input schematic involves a CI to be built). 

You do have the option to buy a midi input board to use, i will need to code support for MIDI input clock to make it work.

But i totatly agree with you, MIDI input is a must have, thats why im working on aciduino V2 to overcome some of the limitations of V1, like MIDI input via USB for easy integration with computer setup like ableton live or whatever daw you're using and also keep it ready for a MIDI input board in case you want to use a native MIDI connection instead of MIDI/USB.

01GOD, 2021-04-03:
Better to make a clone of DELAY LAMA style sounds and add a sequence
randomization section similar to ABL 3.

Doing exact clones of analog on sound on an arduino doesn’t sound like a
fun adventure. Easier to clone digital style

midilab, 2022-10-14:
For now i will leave this request as a future possible implementation on the roadmap. thanks for the request, it always help the project to grow.