midilab/aciduino #7: Quel dossier pour pcb

ptitben313, 2021-10-20:
Bonjour c quel dossier pour faire faire un vrai pcb je suis néophyte donc merci d avance

ijnekenamay, 2021-10-20:
I don't understand French very well, so I apologize if I am misinterpreting you.

Hello. I am a PCB data producer.
In the [PCB] folder of [Hardware], I have published data for KiCAD, a general-purpose PCB CAD system.
You can output the "Gerber" from these and ask a Chinese PCB fabrication company (such as JLCPCB) to manufacture it for you.

If you are not using Kicad, I will add the Gerber files to the PCB folder in the near future.
You can just throw it to the fabricator and get!

midilab, 2021-10-21:
Great! @ijnekenamay  has plans to put the PCB on sales, so it will be easy for those without experience to just grab from him.

Anyway, if you can't wait to start, i think @ijnekenamay  can provide the gerber files like he said before. 

But most of the times the gerber files have some minor details and rules related to a specific fab house for production, so probrably if you want to order your self, its good to use the same fab house as @ijnekenamay has order the first release.