midilab/aciduino #13: Stuck in the programming part

ssid1, 2022-06-26:
I'm really interested in that sequencer but I haven't used arduino before. Now I'm finished the hardware part and don't know what to do next.Could you make a step-list about how to install? Little bit confused because the software part are too many files.

midilab, 2022-06-27:
hey ssid! 

sure, i will workout a install guide on README.md file soon, i will let you know when done.

But in short you need to:

1. Install Arduino environment IDE. 
2. Download latest release of aciduino zip file here: https://github.com/midilab/aciduino/releases
3. Unzip/Untar the release file and using Arduino IDE open the file aciduino-1.2.0/Aciduino/Aciduino.ino
4. With your board connected to PC and configured accorddly on Arduino IDE, press Sent button and wait untill you get a success message that the code was uploaded.

ssid1, 2022-06-27:
I'm using Mac OS system. 

Is it because I connected pin 1 and 2? (I use the PCB broad and solder them all.)

ssid1, 2022-06-28:
No, I don't think there's anything changed after I reset, reconnect and restart the IDE.

midilab, 2022-06-28:
If you get thru all the steps and nothing happens you probably brick or burn your arduino.

Here it goes the list of steps at the link:

Here are some easy things that can often fix this error:

    Disconnect and reconnect the USB cable.
    Press the reset button on the board.
    Restart the Arduino IDE.
    Make sure you select the right board in Tools ► Board ►, e.g. If you are using the Duemilanove 328, select that instead of Duemilanove 128. The board should say what version it is on the microchip.
    Make sure you selected the right port in Tools ► Serial Port ►. One way to figure out which port it is on is by following these steps:
        Disconnect the USB cable.
        Go to Tools ► Serial Port ► and see which ports are listed (e.g. COM4 COM5 COM14).
        Reconnect the USB cable.
        Go back to Tools ► Serial Port ►, and see which port appeared that wasn't there before.
    Make sure digital pins 0 and 1 do not have any parts connected, including any shields.

If none of those work, you will want to try to isolate the issue by replacing things: try a different computer on the same arduino, try a different arduino on the same computer, and try using a different USB cable.

If the issue is with the computer:

    Double-check all computer-related issues in the "easy fixes" list above.
    Reinstall the IDE.
    Reinstall the drivers.

If the issue is with the Arduino:

    Double-check all board-related issues in the "easy fixes" list above.
    Make sure the microcontroller is seated correctly.
    You may need to [burn the bootloader](https://arduino.stackexchange.com/a/474/37).
    Replace the microcontroller if you have another one handy nearby.
    You may have bricked your Arduino. Sorry :(

midilab, 2022-07-05:
Closing by inactivity

closed 2022-07-05 by midilab