midilab/aciduino #17: Midi clock loss

2b2a2t2, 2022-11-15:
Hello, 
using V2B03
I was able to reproduce the issue several times, using different pluggins, tweaking the resonance CC makes the aciduino crash, interface and sequencer are still running, controls are responding, but the DAW says the MIDI connection was lost, and my MIDI monitor does not receive information anymore from the aciduino.

midilab, 2022-11-15:
Any code changes made for this beta3 from your side? like new mici CC or number of tracks?

Are you using teensy lc or 3.6?

thanks for test and reporting!

2b2a2t2, 2022-11-16:
using a Teensy 3.2, the only code changes I made are the button pins (had 2 pairs of buttons reversed)
I no longer have this issue since I uncommented the extra CC in the live page and re-upladed the sketch in the teensy

closed 2022-11-16 by 2b2a2t2

midilab, 2022-11-16:
great! auto fix bugs!
the midi control object part has some flaws for extend that i intent to correct soon, i know it works on the beta but add or remove or creating another track without made some changes could lead to that freeze.

midilab, 2022-11-16:
what are the new cc you are looking for? drums?

2b2a2t2, 2022-11-17:
in the 808 engine, one CC per voice would be great !