midilab/aciduino #19: Pot question... or encoders

2b2a2t2, 2022-11-22:
Could there be a way to get 5-6 pots from V1 to match with on screen parameters ? or even better, encoder so we don't have value "jumps" when changing page/layer/engine ?

midilab, 2022-11-23:
yes thats possible, that will be a module to be released(wiring and code definitions option) to extend the lite basic version.

midilab, 2022-12-02:
two questions:

1- how often are your pot jumping when changing page/layer/engine?

I will write a pickbyvalue option for Pots but im not sure if that will be a good UX choice without testing.

2- Do you think a encoder could be a better option for values and midicc? 

Im using a encoder instead of pot on uone and for general stuffs is quite better specially to avoid jumps, but for sure its a nigthmare to drive midi CC on acid 303 lines. thats the reason i choose a pot instead of encoder for the aciduino lite so people with minimal interface could experiment the acid twists of a pot.

But in case someone is going to use a extension of 4051 for more pots then i think could be a good option to let the user change it easly on the code side(as a #define).

midilab, 2022-12-09:
any news on this one?

closed 2022-12-11 by midilab