midilab/aciduino #24: Aciduino v2 build question

copych, 2022-12-02:
Hello, @midilab ! It was great to know that somebody is making such cool stuff. Some time ago I wanted to build a handy "toy" with real pots and buttons to make some acid fun, like playing 2 basslines along with drums. I thought that I would find something more or less ready, but nope. Okay , guys all over the net have written a lot of DSP code: filters, delays, reverbs etc. So I tried to make it play on two cores of esp32. For the time being I have it working on a esp32 wrover with 4mb psram and an external DAC, and gonna make it available to public. It's midi driven, and could be a perfect pair for the aciduino2 (ok, not perfect, as it's far from 303 + 808 sound yet, but still fun). So the question is: how can I build Aciduino v2 for an Avr or ESP ? I am not a real coder, and #include <functional> makes me scared )

midilab, 2022-12-03:
hey @copych ! 

Im really curious about how it sounds, do you have a video or demo?

Aciduino v2 is based on uCtrl and uClock libraries, so far those libraries are ported to work with AVRs, Teensys and Seedstudio Xiao and the idea is to port to other platforms too, esp32 is not supported rigth now but it can be done. You can open a issue on uClock project asking for a esp32 port, since i dont have one sitting here with me probrably i will need your help tests and debug. 

So bassicly after creating a support for esp32 on [uCtrl](https://github.com/midilab/uCtrl) and [uClock](https://github.com/midilab/uClock) you can use aciduino v2 arduino source code with any of the supported platforms(that has the minimun epprom and ram for aciduino v2 firmware).

Did you try any teensy >= 3? they have a nice audio library to create synths and i see a lot of guys out there doing some amazing sounding stuffs with it, maybe port your work for teensy can be a short path to the aciduino + sound engine?
There is a sample support too on teensy audio library too beside the really nice DSP stuff, so i plan to get a engine for sampler to be used along with 808 for those who need a standalone solution and later i got this TODO list to check how far i can go with teensy audio library dsp stuff to get a 303 sound from it.

copych, 2022-12-03:
Thanks for the reply, @midilab !
I'll try to record some sounds from the breadboard =)
As of Teensy, I am sure it's a great piece of gear, but only starting with v4.0 (600MHz), cause AFAIK v3.+ only gives 120MHz single core, which is less than ESP32 which has (2 x 240MHz cores). But looking at Teensy 4.+ we have price level ($70+) of a card-sized PC (OrangePI or similar) running @GHz.. So taking the price into account, I choose ESP32 ($10).
About the libraries, you have mentioned that AVR is supported, right? Should I just use library manager in Arduino IDE?

copych, 2022-12-03:
https://www.dropbox.com/s/tf3pz1pt6qpfezv/demo.wma?dl=0

copych, 2022-12-03:
https://github.com/copych/AcidBox uploaded src

midilab, 2022-12-03:
> Thanks for the reply, @midilab ! I'll try to record some sounds from the breadboard =) As of Teensy, I am sure it's a great piece of gear, but only starting with v4.0 (600MHz), cause AFAIK v3.+ only [...]

Actually teensy 4 is $23 dolars, kinda 2x the price of esp32 and 6.5x more powerfull for processing, so i guess its a winner for real horse power for DSP. Its a huge jump from 32bits to 64bits arch and it also contains a FPU and DSP extension embedded. Just imagine that the M8 tracker is based on a single teensy 4 then you can picture how low cost it is for the power it has for DSP.

[Teensy coremark](https://www.pjrc.com/store/coremark_barchart_t40_small.png)
[Teensy4](https://www.pjrc.com/store/teensy40.html)

Any way you got a point for 10 dolars ARM being a real low cost too. But teensy audio library can save you a lot of work on DSP side that on a long run project needs to be acountable too.

For uClock you can use Arduino library manager to install it from github, but the better current way of using is to put all libs inside /src folder of your arduino project, jsut following the AciduinoV2/src strcuture you will be good to go.

> https://www.dropbox.com/s/tf3pz1pt6qpfezv/demo.wma?dl=0 looping, mouse control 1cc at a time. )

The square wave is convincing! can you twist some filter around so i can get an idea of the filter quality.

> https://github.com/copych/AcidBox uploaded src

Hey great job! unfortally i dont have a esp32 to test it out. Open a issue as request for new platform esp32 to be supported on uClock then we have a start point to check if we can merge the 2 acid projects into a standalone diy acid box.

copych, 2022-12-04:
Thanks for the info! I'd surely prefer a Teensy for the price indicated, but unfortunately, the world is not ok lately, and I can't pay directly to the US, as well as they are probably not delivering to Russia. Buying from China or from 3rd party gives the price I've mentioned. About the filter, I'm not quite satisfied yet, and now experimenting with the one from the open303 bundle. Also there's one more in SoundPipe collection to test. Thanks again for the attention

midilab, 2022-12-05:
You got a point! I got same issues here, buy anything directly from US will cost me something around 3x the original price, the only way out is always to DIY as much as possible. 

So in that way ESPs are avaliable for good prices all over the world, so its good to see projects like yours making use to the arm power to DSP some opensource noises and notes.

It would be great to follow the progress of your project with the filter research.

Open the issue i mention, there is already one opened for a new platform request, so maybe you can take a read to follow up with the help i need to port uClock and uCtrl to ESPs on the new issue.

copych, 2022-12-05:
Getting back to the matter, when I tried to compile in ArduinoIDE for Leonardo, I got the following
```
In file included from D:\xxx\arduino\AciduinoV2\src\uCtrl\uCtrl.h:75:0,
                 from D:\xxx\arduino\AciduinoV2\AciduinoV2.ino:39:
D:\xxx\arduino\AciduinoV2\src\uCtrl\module\page\page.hpp:33:10: fatal error: functional: No such file or directory
 #include <functional>
          ^~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Leonardo.
```

midilab, 2022-12-05:
uCtrl needs a review for AVR since i added some new stuffs while working on aciduino teensy version, i will try to do it asap. It looks like functional is not available for AVRs, you can try to find the correlated include for this one to compile.

But in the mean time i can teel you that leonardo doesn't have the minimum ram and epprom to run aciduino v2.
I think the only AVR that can handle it is Mega2560.

midilab, 2022-12-05:
you also have the option to use aciduino v1 wich is can run on any AVR arduino based. but its only 303 sequencer with no oled screen.

copych, 2022-12-05:
Still trying to compile for AVR
```
name.c:5:10: fatal error: usb_names.h: No such file or directory
 #include "usb_names.h"
          ^~~~~~~~~~~~~
compilation terminated.
exit status 1
usb_names.h: No such file or directory

```

midilab, 2022-12-05:
you can delete name.c since its a teensy only usb midi driver initiation.

copych, 2022-12-05:
Oh I've got several pages of error output, you'd better look at it with your own eyes. I think that would be a bad idea to scroll all this output in here)

midilab, 2022-12-05:
you can create a gist for that i guess

midilab, 2022-12-05:
you need to comment the USE_USB_MIDI too, its a teensy option only, and ucomment USE_UART_MIDI

copych, 2022-12-05:
https://gist.github.com/copych/0ff0b755c5c8aa3fe3fcd09da66f5205
tried to disable both USE_USB_MIDI and USE_UART_MIDI, target is set to Mega

midilab, 2022-12-05:
Uhmmm..

Try to use this function(replace per the one on acid_ui_components.ino):

void genericOptionView(const char * title_char, String value, uint8_t line, uint8_t col, bool selected, bool large = false)
{
  String title(title_char);
  const uint8_t view_size = 12;
  uint8_t x=(col-1)*5, y=(line-1)*8;
  uint8_t value_col = large ? (view_size*2)-value.length()+1 : view_size-value.length();
  uCtrl.oled->display->drawBox(x+1, y, 1, 7);
  uCtrl.oled->print(title, line, col+1, selected);
  uCtrl.oled->print(value, line, col+value_col, selected);
}

copych, 2022-12-05:
https://gist.github.com/copych/c31840bc3af368172183562083dfe2ad

midilab, 2022-12-05:
doesn't look like those compile erros are using the function above

midilab, 2022-12-05:
One thing that could be done from your side until i push the changes is to replace all genericOptionView() calls over the code base changing the second argument to be passed to a String object before genericOptionView(somethign like change value_x per String(value_x), so it converts before goes to the function.

copych, 2022-12-06:
C:\Users\admin\AppData\Local\Temp\cccFAY5h.ltrans0.ltrans.o: In function `freeram':
E:\YandexDisk\arduino\AciduinoV2/page_welcome.ino:7: undefined reference to `sbrk'
collect2.exe: error: ld returned 1 exit status
exit status 1

midilab, 2022-12-06:
you can comment freeram() function and all references too, its a teensy function only.

midilab, 2022-12-14:
i will release a multi-arch version soon to support either teensy and avrs(with minimun memory specs).

do you have a video or audio of the current acidjukebox? looks interesting! you're about right, never hear about before this acid-banger.

copych, 2022-12-14:
i've just put a demo on the main page of the AcidBox repo. No buttons pressed, no knobs turned.

midilab, 2022-12-21:
> i've just put a demo on the main page of the AcidBox repo. No buttons pressed, no knobs turned.

Sounds promisor!

You can now build out of the box aciduino v2 for mega 2560 o beta 7 release:
https://github.com/midilab/aciduino/releases/tag/v2.0.0-beta7

[image]

copych, 2022-12-21:
Oh! Nice that the project moves on! I hope someday it'd be ported to ESP32 ) because of arduinos I only have ProMicro which is identical to Leonardo, but it won't allow to run your app (

midilab, 2022-12-21:
soon i guess, its on todo list already.

closed 2022-12-21 by midilab

estufilla, 2023-11-18:
> you also have the option to use aciduino v1 wich is can run on any AVR arduino based. but its only 303 sequencer with no oled screen.

Is there any way that this can be explained? Been trying for 3 days straight to use a Uno or Leonardo (or Pro Micro) with Aciduino V1, but have been just literally impossible for me.

midilab, 2023-11-18:
The code is ready for arduino Nano, other AVR microcontrollers should work but maybe it needs some changes on code side to the specific arduino model, i cant tell since i've never try, one the things could be messing up its the EPPROM size differences on nano and leonardo, those part of the code needs to be ported to run fine. anyway, its on todo list, get leonardo running on acidunio v1 with usb midi support.