midilab/aciduino #28: System Freeze in Seqr Page 2

2b2a2t2, 2022-12-27:
had the problem with every beta so far...
if the src/sequencer/setup, when the pattern length is modified for something above 64 steps, the second page UI is not designed for such a long pattern and makes the teensy unresponsive. Could a security implementation prevent from switching to the second page ?

midilab, 2022-12-29:
prevent to access the page is a bad UI/UX behaviour, a good solution will be finish the step grid to support more than 64 steps.

i will see what i can do for that matter.

midilab, 2022-12-30:
Any way until some changes go for release you can do that on page_sequencer.ino inside 
...
} else if (subpage == 1) {
    //...
    //... 
    if (!AcidSequencer.is303(_selected_track))
        uCtrl.page->component(stepSequencerComponent, 2, 1, true);
}

that way the stepsequencer will only showup as full size if its not a 808 track...

midilab, 2023-01-02:
implemented a support for up to 256 steps on 303 for UI.

https://github.com/midilab/aciduino/releases/tag/v2.0.0-beta9

closed 2023-01-02 by midilab