midilab/aciduino #30: ESP32s2

copych, 2023-01-13:
Hello, @midilab , I'm trying now to run it on the ESP32S2 board, how do I set pins for OLED SDA and SCL?
Please, note that ESP32, ESP32S2, ESP32S3 and ESP32C3 are VERY different boards. They need different settings for cores and for pins...
Say, ESP32S2 only has just one core, has no FPU, has WiFi, but no BT, has just 2 UARTS, but has native USB support, including USB-OTG, while ESP32 has 2 cores, has FPU, has 3 UARTS, has WiFi+BT, but no built-in USB etc...

midilab, 2023-01-13:
Since its experimental support, im counting on the community to help testing, document or come up with solutions. 

So in case you have more than one board for example and want to test it, you can document everything related to the port so we can add to the project(for example the tests you have done for the GPIOs i've updated at the schematics, but its always welcome an effort to pull changes on code or schematics when they come up).

By now everything related to drivers and devices are setup inside 1_uctrl.ino.

`//
// Display device
//
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8X8_PIN_NONE);
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);
`
So you just need to comply with u8g2 driver usage and pinage setup. the uCtrl just uses the object of device instead of trying to configure it self, leading to the problem of a reduced portability.

copych, 2023-01-13:
ok, i'm just not familiar with u8g2 drivers, and esp32s2 seems not to have default i2c pins, so i need to assign them explicitly, that's why i ask.

midilab, 2023-01-13:
you can check they reference manual for that, i never had to change the default i2c before:
https://github.com/olikraus/u8g2/wiki/setup_tutorial

bassicly you just need to follow the manual and change the line for that on 1_uctrl.ino.

If you succeed, please create a pull request with the changes so i can add to the the codebase.

Ideally we should create a schematic for each of ESP32XXXX since it looks like we have different boards pinage. If you could copy the [one we have](https://github.com/midilab/aciduino/blob/master/v2/hardware/aciduino_v2-esp32.fzz) and make the modifications needed and rename it to the board name - like aciduino_v2-esp32s2.fzz - and pull that will be a great contribution. After that we need to find the FLAGS that identify each of these boards to make the code compile transparent for all boards we test so far.

copych, 2023-01-14:
Thanks for the link, I've managed to launch tha app on a ESP32-S2 (LOLIN S2 MINI), I'll send what I have later

closed 2023-01-23 by midilab