midilab/aciduino #31: Nano

jaseat1, 2023-01-25:
hi, I'm really interested in building this, but would be great if it could run on a Nano instead of the mega. Is this possible?

jaseat1, 2023-01-25:
Thank you for the fast reply. Dam!I was hoping it would work on a nano, just to keep the project small, also only have nanos at the moment. Just makes a nicer project to have the screen aswell. Thank you.

midilab, 2023-01-25:
No problem at all! 

For small option you can go for teensy 3.2 or 4.0 or even a ESP32. unfortally the only AVR avaliable so far that has the minimun flash and ram is Mega.

jaseat1, 2023-01-25:
I guess it would also work on a mega mini as they share the same chipset?

midilab, 2023-01-25:
Yes it should run fine! Its the same hardware in small size. 

If it happens to you build it on top of it, please report back so i can add officially tested and supported by aciduino too.

closed 2023-01-26 by midilab