midilab/aciduino #34: Encoder acceleration

2b2a2t2, 2023-02-07:
Hello !
Having a lot of fun with my aciduino, I wanted to upgrade. I found 2*4051, I built a Pot16 module and selected
#define USE_CUSTOM_PINOUT
//#define USE_PROTOBOARD_PINOUT

changed the modules.h  page
 //#define USE_CHANGER_POT
  #define USE_CHANGER_ENCODER

uncommented 
#define USE_POT_16      

i get data from the pots, but if I disconnect the pot16 module, all 16 MIDI CC values get crazy (because the muxs are not connnected).
So i went back to the code and commented //#define USE_POT_16... 

Could there be an option so we can switch pot16 on/off onboard ?
Also I now have both CHANGER_POT and CHANGER_ENCODER hooked on the device since the encoder is not really usable when the pot16 is not connected, it is much more convenient when tweaking options but MIDI CCs are so slow it reminds me of an Arturia's controller ;p !
Could you implement an acceleration setting for the encoder ?

PS : Also something weird : "Aciduino" is now "Teensy MIDI"... and I promise I didn't change name.c

midilab, 2023-02-09:
> Could there be an option so we can switch pot16 on/off onboard ?

What are your needs? how often do you connect and disconnect them?

> Also I now have both CHANGER_POT and CHANGER_ENCODER hooked on the device since the encoder is not really usable when the pot16 is not connected, it is much more convenient when tweaking options but MIDI CCs are so slow it reminds me of an Arturia's controller ;p !

That was the main reason behind a pot instead a encoder on the minimal version of aciduino v2, no matter the quality of your encoders, they will never ever going to behave like the old and good pot for midi cc changes and i really think they are unusable for the matter of midi cc, but the best for value changing.

I dont know if that will work, but did you try to enable both CHANGER_POT and CHANGER_ENCODER so you can make use of the most appropriated one when needed?

> Could you implement an acceleration setting for the encoder ?

I was expecting someone to complain about the speed, most machines i see fo far make use of shift to acceletare the pot change, could be an option for aciduino but there are places where shift+pot is used too so i need to check a good way of doing this, and i agree, its a needed to have the option to acceletare it.

> PS : Also something weird : "Aciduino" is now "Teensy MIDI"... and I promise I didn't change name.c

No idea about, but i guess thats something related to your side of teensyduino compiler setup(IDE update... midi options changed?)

2b2a2t2, 2023-02-14:
> What are your needs? how often do you connect and disconnect them?
well sometimes the aciduino is connected to the computer, sometimes it is connected to the phone, if I take the aciduino out for a quick jam on the phone I don't take the pots... ;p

> no matter the quality of your encoders, they will never ever going to behave like the old and good pot for midi cc changes
you could definitely try a midi fighter twister, feels just like a "real" pot, the code is on github https://github.com/DJ-TechTools/Midi_Fighter_Twister_Open_Source
(but they use a M74HC165M1R chip to count each encoder)
also take a look at Zynthian way of implementing encoders... their acceleration detection is pretty efficient and smooth (but they added some capacitors lately so they also may have changed the code)

midilab, 2023-02-19:
> you could definitely try a midi fighter twister, feels just like a "real" pot, the code is on github https://github.com/DJ-TechTools/Midi_Fighter_Twister_Open_Source
(but they use a M74HC165M1R chip to count each encoder)
also take a look at Zynthian way of implementing encoders... their acceleration detection is pretty efficient and smooth (but they added some capacitors lately so they also may have changed the code)

Thanks for the references! Good to know some devices that feels ok for Potting with a encoder. 

uCtrl makes use of 74H165 too, but thats not the point for velocity or smoothness of changes. Its related mainly to the speed of reading wich takes use of cpu and it is running inside a ISR sharing processing time in a multiplexed kinda of way with other drivers, so we have a sort of limit(not explored yet) of how smooth changes can be using uCtrl. 

Thats also related to the quality and resolution of the encoder, i dont have a good resolution one with me to test, but maybe we can try to figure out how to do it if you have one around and could test it.

> well sometimes the aciduino is connected to the computer, sometimes it is connected to the phone, if I take the aciduino out for a quick jam on the phone I don't take the pots... ;p

Originally i wrote the uCtrl and compose the uMODULAR modules to be able to be able to hotswap modules, so i can try to think of a way to enable that in a user friendly way for the next releases.

2b2a2t2, 2023-08-03:
I can test whatever code you want ! I also can provide the code I'm using if you wan't to give an eye...

midilab, 2023-08-03:
great! please show me the code you have tested so far and i will take a look at. 

So this code you're using, how does it feels about the user interaction compared to the other gears you have and does good acceleration?

midilab, 2023-09-24:
closing issue and move it to [uCtrl](https://github.com/midilab/uCtrl/) roadmap.

closed 2023-09-24 by midilab