midilab/aciduino #37: Compilation errors for Teensy 3.2

espcam, 2023-06-22:
Hi!
I'm filing issue after discussing it with author. 
I tried to verify the code in Arduino IDE 1.8.19, Teensyduino 1.58, Teensy 3.2 board - it outputs the same errors no matter if i switch Serial, MIDI, MIDI + Serial modes. I used master branch, then Aciduino v2 rc1 - no big difference. 
Main PC - Fedora 37, Kernel: 6.3.8-100.fc37.x86_64 , Arduino 1.8.19, Teensyduino 1.58. (compilation logs attached)
Windows laptop - got the same errors on Windows 8.1, Arduino 1.8.19, Teensyduino 1.58. (windows log attached)

[error_midi_mode.log](https://github.com/midilab/aciduino/files/11834674/error_midi_mode.log)
[error_midi-serial_mode.log](https://github.com/midilab/aciduino/files/11834675/error_midi-serial_mode.log)
[windows8.1_compilation.log](https://github.com/midilab/aciduino/files/11834676/windows8.1_compilation.log)

Steps to reproduce: open v2 rc1 project, connect the board and choose MIDI or MIDI + Serial USB mode, Teensy board is detected by default, choose Teensy port and press "Verify" button in Arduino IDE. 

I'm ready to provide extra info if needed.

midilab, 2023-06-22:
Hi @espcam 

Thanks for reporting and information about it.

Looking at the logs something really strange is happen, the compiler is trying to compile using AVR libraries of teensy instead of ARM ones.

`Using library Wire at version 1.0 in folder: /home/user/arduino-1.8.19/hardware/teensy/avr/libraries/Wire `

As you see looks like there is a problem with teensyduino libraries, as it is using AVR instead of ARM even if you select teensy 3.1/3.2 that is a ARM one.

So i would like to ask you to check your teensyduino installation and try to uptade to 1.56 that im aware of working well. then we can follow from there.

I'm using 1.56, but there is newer versions already, so please try to compile using the same as mine, if it works update to the latest and please tell me if it works either. so i can write at the docs the tested versions for correct compilation.

espcam, 2023-06-22:
Downloaded Arduino 1.8.19 again and installed Teensyduino 1.56 - Compilation succeeded.
I also tried to compile project for ESP32-S3, but had no luck on either versions of software. I'll attach error log for ESP32-S3 here for future research. My issue can be closed i guess, as I believe 1.58 will be also supported later on. 
[ESP32-S3_error.zip](https://github.com/midilab/aciduino/files/11835906/ESP32-S3_error.zip)

closed 2023-06-22 by espcam

midilab, 2023-06-22:
Great that it works! hope you have some acid fun with it.

By the way, so far i have only enable support for [this ESP32 board](https://github.com/midilab/aciduino/blob/master/v2/hardware/imgs/aciduino_v2-esp32_bb.png), so i cant tell if that will compile for others EPS32 for now.

But i really want to get both projects Acidbox and Aciduino on a single ESP32, as soon as i get some free time to work on it.