midilab/aciduino #38: unable to compile with the Arduino-IDE

Di3g0, 2023-06-27:
testet on archlinux 
arduino-ide_2.0.4_Linux_64bit.AppImage:
`
/home/diego/develop/aciduino/v2/AciduinoV2/AciduinoV2/AciduinoV2.ino:61:10: fatal error: src/uClock/src/uClock.h: No such file or directory
   61 | #include "src/uClock/src/uClock.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: src/uClock/src/uClock.h: No such file or directory
`

How to reproduce:
phuture@acidlab:~$ git clone https://github.com/midilab/aciduino.git
phuture@acidlab:~$ cd aciduino/
phuture@acidlab:~$ git submodule update --init --recursive

then open the main file from "AciduinoV2.ino" with the arduino-ide

midilab, 2023-06-28:
Hi @Di3g0,

Thanks for the report.

Looking at the logs show me that you have a folder "AciduinoV2" inside the folder "AciduinoV2" wich is not the orignal folder structure of the project.

Try to delete your local aciduino repository and clone it again with submodule update.

If you're still getting problems with cloned version of repository you may try to use the [release pack](https://github.com/midilab/aciduino/releases/download/v2.0.0-rc1/aciduino-v2_rc1.zip) instead for compilation.

Di3g0, 2023-07-05:
sorry for the late reply.
I have tried your release pack but unfortunately had no success maybe you have a tip for me. I have attached the compiler errors as a file
[compiler-error.log](https://github.com/midilab/aciduino/files/11962428/compiler-error.log)

midilab, 2023-07-06:
What is the teensy model you are trying to compile for?

Di3g0, 2023-07-31:
hey, I'm just curious and wanted to ask you if you have already found a solution or workaround to this problem

midilab, 2023-08-01:
> What is the teensy model you are trying to compile for?
> What is the Teensyduino version you are using?

Since i couldn/t reproduce the problem im still waiting from your input to continue with the debug.

Di3g0, 2023-08-01:
Also tested now with Windows 10 x64 and a fresh Arduino 2.1.1 installation 

OS:          Windows 10 ( fresh VM )/ Arch Linux 64
Arduino:  Arduino 2.1.1
Board:      Teensy 3.2

teensy:teensy-tools@1.58.0
teensy:teensy-compile@11.3.1
teensy:teensy-discovery@1.58.0
teensy:teensy-monitor@1.58.0
teensy:avr@1.58.1

on both setups I get the same error if i try to compile from the source 
I have some time this weekend and will have a look at the compiler since your code doesn't seem to give any real errors but only warnings.
I'm still surprised that only my setup is affected by it. 

this seems to be the crucial point
`
c:/users/diego/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld.exe: c:/users/diego/appdata/local/arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/thumb/v7e-m/nofp\libc.a(libc_a-writer.o): in function `_write_r':
(.text._write_r+0x14): undefined reference to `_write'
collect2.exe: error: ld returned 1 exit status
`

Interplain, 2023-08-05:
Hi, Thanks for making this possible. Just as Di3g0 suggests i have the same issue. I tried with Arduino IDE 2.1.1 and Teensy 3.0 with the Release Pack i managed to install the 1.56.1 but can seem to select it from the board manager. Tried an older version of IDE 1.8.16  It compiles and shows the Hex uploader:
Opening Teensy Loader...
Sketch uses 92728 bytes (70%) of program storage space. Maximum is 131072 bytes.
Global variables use 9916 bytes (60%) of dynamic memory, leaving 6468 bytes for local variables. Maximum is 16384 bytes.
But has a lot of Orange text relating to errors:
warning: no return statement in function returning non-void [-Wreturn-type] many times relating to (engine.h)
would love to get this running.

midilab, 2023-08-05:
Hi @Interplain ,

`managed to install the 1.56.1 but can seem to select it from the board manager.`
Arduino 2 will only use the last installed version, it only shows to you the versions installed when you go to the screen to select a new one to install.

The problem is related to supported version, since arduino 2 and teensyduino evolute we will need check for break changes and other things. I will put on roadmap the arduino 2 support with new releases.

`would love to get this running.`
Looks like you already have, since you compile without errors(dont mind about the warnings). You can go and try your compiled version with arduino 1.8 and teensyduino 1.56(wich now is the supported versions of aciduino).

Please let me know about!

Interplain, 2023-08-05:
That's great, thanks for the reply. I tried it with the ESP-Vroom-32 but it failed to compile. So i have a Teensy 4.0 on its way today. Thing is i have all the part and tried with the Vroom board. Confident it will work now you told me that the Orange warnings are nothing to worry about.

midilab, 2023-08-05:
Do you mind to share the compile errors of ESP32? also all the setup you made with versions and so on?

So far i have release a experimental support for those boards but no one had test so far to my knowledge, i was able to compile and run the interface without problems but no extensive use to know if it fully works as expected.
here it is the schematics for that board: https://raw.githubusercontent.com/midilab/aciduino/master/v2/hardware/imgs/aciduino_v2-esp32_bb.png

Interplain, 2023-08-05:
Yeah sure, give me a few minutes. Would like help you in what way i can. my email is Interplain1@gmail.com can you delete. after reading. i dont want my email everywhere. https://soundcloud.com/interplain

Interplain, 2023-08-05:
Thank you for the reply. OK, using the DOIT ESP32 DEVKIT V1 in the
Aurduino IDE 2.0

I compile and I get this.

C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino: In function
'void uCtrlSetup()':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:136:19: error:
'NAV_SHIFT_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_SHIFT_PIN);
                   ^~~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:145:19: error:
'NAV_FUNCTION1_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_FUNCTION1_PIN);
                   ^~~~~~~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:147:19: error:
'NAV_FUNCTION2_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_FUNCTION2_PIN);
                   ^~~~~~~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:149:19: error:
'NAV_GENERAL1_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_GENERAL1_PIN);
                   ^~~~~~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:151:19: error:
'NAV_GENERAL2_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_GENERAL2_PIN);
                   ^~~~~~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:153:19: error:
'NAV_RIGHT_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_RIGHT_PIN);
                   ^~~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:155:19: error:
'NAV_UP_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_UP_PIN);
                   ^~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:155:19: note:
suggested alternative: 'NOT_A_PIN'
   uCtrl.din->plug(NAV_UP_PIN);
                   ^~~~~~~~~~
                   NOT_A_PIN
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:157:19: error:
'NAV_DOWN_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_DOWN_PIN);
                   ^~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:157:19: note:
suggested alternative: 'NOT_A_PIN'
   uCtrl.din->plug(NAV_DOWN_PIN);
                   ^~~~~~~~~~~~
                   NOT_A_PIN
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:159:19: error:
'NAV_LEFT_PIN' was not declared in this scope
   uCtrl.din->plug(NAV_LEFT_PIN);
                   ^~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:159:19: note:
suggested alternative: 'NOT_A_PIN'
   uCtrl.din->plug(NAV_LEFT_PIN);
                   ^~~~~~~~~~~~
                   NOT_A_PIN
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:217:19: error:
'CHANGER_POT_PIN' was not declared in this scope
   uCtrl.ain->plug(CHANGER_POT_PIN);
                   ^~~~~~~~~~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:217:19: note:
suggested alternative: 'CAN_BRP_MIN'
   uCtrl.ain->plug(CHANGER_POT_PIN);
                   ^~~~~~~~~~~~~~~
                   CAN_BRP_MIN
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\acid_ui_components.ino: In
member function 'virtual void TopBar::view()':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\acid_ui_components.ino:212:29:
[... 89 lines cut ...]
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h:42:35:
note: in definition of macro 'ATOMIC'
 #define ATOMIC(X) noInterrupts(); X; interrupts();
                                   ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\midi.ino: In function 'void
midiInputHandle()':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\midi.ino:119:10: error:
'usbMIDI' was not declared in this scope
   while (usbMIDI.read()) {
          ^~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\page_welcome.ino: In
function 'void welcome_page_refresh(uint8_t)':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\page_welcome.ino:41:28:
error: 'freeRam' was not declared in this scope
   uCtrl.oled->print(String(freeRam()), 3, 11);
                            ^~~~~~~
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\page_welcome.ino:41:28:
note: suggested alternative: 'fread'
   uCtrl.oled->print(String(freeRam()), 3, 11);
                            ^~~~~~~
                            fread

exit status 1

Compilation error: 'NAV_SHIFT_PIN' was not declared in this scope

midilab, 2023-08-05:
Can you try with arduino 1.8.18? until now its the official supported version.

But thanks for the logs, it will be usefull when start to port aciduino to arduino 2.0.

Interplain, 2023-08-05:
Ok, I installed 1.18.1 IDE and tried to compile it with the same board
selected DOIT ESP32 DEVKIT V1.
Arduino: 1.8.18 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600,
None, Disabled"

C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino: In function
'void uCtrlSetup()':

1_uctrl:136:19: error: 'NAV_SHIFT_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_SHIFT_PIN);

                   ^~~~~~~~~~~~~

1_uctrl:145:19: error: 'NAV_FUNCTION1_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_FUNCTION1_PIN);

                   ^~~~~~~~~~~~~~~~~

1_uctrl:147:19: error: 'NAV_FUNCTION2_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_FUNCTION2_PIN);

                   ^~~~~~~~~~~~~~~~~

1_uctrl:149:19: error: 'NAV_GENERAL1_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_GENERAL1_PIN);

                   ^~~~~~~~~~~~~~~~

1_uctrl:151:19: error: 'NAV_GENERAL2_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_GENERAL2_PIN);

                   ^~~~~~~~~~~~~~~~

1_uctrl:153:19: error: 'NAV_RIGHT_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_RIGHT_PIN);

                   ^~~~~~~~~~~~~

1_uctrl:155:19: error: 'NAV_UP_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_UP_PIN);

                   ^~~~~~~~~~

C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:155:19: note:
suggested alternative: 'NOT_A_PIN'

   uCtrl.din->plug(NAV_UP_PIN);

                   ^~~~~~~~~~

                   NOT_A_PIN

1_uctrl:157:19: error: 'NAV_DOWN_PIN' was not declared in this scope

   uCtrl.din->plug(NAV_DOWN_PIN);

                   ^~~~~~~~~~~~

C:\Users\inter\Documents\Arduino\v2\AciduinoV2\1_uctrl.ino:157:19: note:
suggested alternative: 'NOT_A_PIN'

   uCtrl.din->plug(NAV_DOWN_PIN);

[... 187 lines cut ...]
   while (usbMIDI.read()) {

          ^~~~~~~

C:\Users\inter\Documents\Arduino\v2\AciduinoV2\page_welcome.ino: In
function 'void welcome_page_refresh(uint8_t)':

page_welcome:41:28: error: 'freeRam' was not declared in this scope

   uCtrl.oled->print(String(freeRam()), 3, 11);

                            ^~~~~~~

C:\Users\inter\Documents\Arduino\v2\AciduinoV2\page_welcome.ino:41:28:
note: suggested alternative: 'fread'

   uCtrl.oled->print(String(freeRam()), 3, 11);

                            ^~~~~~~

                            fread

exit status 1

'NAV_SHIFT_PIN' was not declared in this scope

midilab, 2023-08-05:
Thanks for the info:

So the problem is that since i've port to esp32 aciduino get this new feature of save patterns and config into epprom, so i will need to adjust it for esp32 too, i think maybe we can get a working version, i will keep you update when i commit the fix for this one.

Interplain, 2023-08-05:
Hope this helps,  Arduino IDE Version 2.1.1 and Teensy version 3.0 selected
from the drop down.  Compiles and opens up the Hex uploader with the
version you last sent AciduinoV 2b08. It compiles also on Teensy 3.2 3.1
but I had noticed that the memory usage changes. I read somewhere that
Teensy 4.0 has a different memory configuration so maybe it is an
ongoing thing.  I compile AciduinoV 2b08 on both IDE 2.1.1 and 1.8.18 and
used Teensy 3.0 on both and the memory usage is the same.
The output from IDE 2.1.1 and 1.8.18 is the same as below.

In file included from
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine_303.h:33:0,
                 from
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\acid_sequencer.h:36,
                 from
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\AciduinoV2.ino:71:
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h: In
member function 'virtual uint8_t Engine::getTrackLength(uint8_t)':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h:64:52:
warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint8_t getTrackLength(uint8_t track) {}
                                                    ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h: In
member function 'virtual bool Engine::stepOn(uint8_t, uint8_t)':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h:68:55:
warning: no return statement in function returning non-void [-Wreturn-type]
     virtual bool stepOn(uint8_t track, uint8_t step) {}
                                                       ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h: In
member function 'virtual uint8_t Engine::getStepData(uint8_t, uint8_t)':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h:72:63:
warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint8_t getStepData(uint8_t track, uint8_t step) {}
                                                               ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h: In
member function 'virtual uint8_t Engine::getCurrentStep(uint8_t)':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h:74:52:
warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint8_t getCurrentStep(uint8_t track) {}
                                                    ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h: In
member function 'virtual uint8_t Engine::getTrackVoice(uint8_t)':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h:78:51:
warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint8_t getTrackVoice(uint8_t track) {}
                                                   ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h: In
member function 'virtual const char* Engine::getTrackVoiceName(uint8_t,
uint8_t)':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\sequencer\engine.h:79:83:
warning: no return statement in function returning non-void [-Wreturn-type]
     virtual const char * getTrackVoiceName(uint8_t track = 0, uint8_t
voice = 0) {}

       ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\acid_ui_components.ino: In
member function 'virtual void TopBar::view()':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\acid_ui_components.ino:212:29:
warning: invalid conversion from 'const uint8_t* {aka const unsigned
char*}' to 'uint8_t* {aka unsigned char*}' [-fpermissive]
           uCtrl.oled->print(SUBPAGE_SELECTED, 1, i+9);
                             ^
In file included from
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\uCtrl\uCtrl.h:41:0,
                 from
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\AciduinoV2.ino:66:
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\uCtrl\module\oled\oled.hpp:62:14:
note:   initializing argument 1 of 'void
uctrl::module::Oled::print(uint8_t*, uint8_t, uint8_t, bool)'
         void print(uint8_t * bitmap8, uint8_t line, uint8_t col, bool
do_blink = false);
[... 391 lines cut ...]
uctrl::uCtrlClass::initAin(uint8_t, uint8_t, uint8_t, uint8_t)'
[-fpermissive]
 bool uCtrlClass::initAin(uint8_t pin1 = 0, uint8_t pin2 = 0, uint8_t pin3
= 0, uint8_t pin4 = 0)

                    ^
In file included from
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\uCtrl\uCtrl.cpp:29:0:
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\uCtrl\uCtrl.h:160:7:
note: previous specification in 'bool uctrl::uCtrlClass::initAin(uint8_t,
uint8_t, uint8_t, uint8_t)' here
  bool initAin(uint8_t pin1 = 0, uint8_t pin2 = 0, uint8_t pin3 = 0,
uint8_t pin4 = 0);
       ^
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\uCtrl\uCtrl.cpp: In
member function 'void uctrl::uCtrlClass::run()':
C:\Users\inter\Documents\Arduino\v2\AciduinoV2\src\uCtrl\uCtrl.cpp:411:10:
warning: variable 'port_ref' set but not used [-Wunused-but-set-variable]
  uint8_t port_ref = 0;
          ^
Opening Teensy Loader...
Sketch uses 78564 bytes (59%) of program storage space. Maximum is 131072
bytes.
Global variables use 10208 bytes (62%) of dynamic memory, leaving 6176
bytes for local variables. Maximum is 16384 bytes.

midilab, 2023-08-05:
@Interplain great!

But the 2b08 was only a reference if you want to use it for teensy LC only, for any other teensy better stick with version 2rc1, the latest: https://github.com/midilab/aciduino/releases/download/v2.0.0-rc1/aciduino-v2_rc1.zip

Interplain, 2023-08-05:
Thanks, I'll check it out! I have the Teensy 4.0 I flashed it with the
version.2b08. It's flashed OK with 1.8.18 and Teensy 4.0 selected in the
Board Manager. Rebooted it and loaded up Ableton and it was found in the
preferences.  I will try the version you just given me. I am in the process
of building the board. I was using Bread Board and using your fritz image.
But the parts a falling out of the board and intermittent connections. So I
started soldering the parts on a green board with holes in it. So I need to
study the wiring closely.  I am under the impression that the Teensy lite
v2 is the correct one to wire it with. The screen I am waiting for same
size has yours. I only have a 0.96. Daft question,  where does the sound
come from? Is it routed through Ableton? Just a few questions I have. I
also have a 3d printer I build so plan to make a custom case.

midilab, 2023-08-05:
Great that works! 

`also have a 3d printer I build so plan to make a custom case.`
Nice, let us know about this case!

` where does the sound come from? Is it routed through Ableton?`
Aciduino is a MIDI sequencer(can be extended to CV/Gate for modular too), wich means it can be used to play instruments inside Ableton or any other DAW with audio plugins that receives midi, or you can use it with external synthesizers and drummachines like those clones of 303 we have seeing nowdays. So no sound comes from him! we got an ongoing idea to integrate other ESP32 project with sound module for drums and basses.

Interplain, 2023-08-05:
Great! I will try it with my synths. That should be interesting. I did once
use a ESP32 with a DAC made a drum machine to trigger TR808 sound. I still
have the source for that.

midilab, 2023-08-06:
`Great! I will try it with my synths. That should be interesting.`
This is i build one for my self, my main usage is to control synths and drum machines and be master clock for the rest of setup, like ableton live in case, its great for live situations.

`I did once use a ESP32 with a DAC made a drum machine to trigger TR808 sound. I still have the source for that.`
Sounds interesting.... here is a guy who have done a [built of 2x 303 and a 808 in a esp32](https://github.com/copych/AcidBox), he was the one who asks for a aciduino port of esp32 some months ago, i will try to build one for my self and later try to integrate aciduino with it.

By the way i have fix the compilation for esp32 if you want to give a try, i didn't have time to test but it compiles fine using arduino 1.8.x and the latest esp32 version(2.11 i guess).

I will have to use the last git commit for that:

```console
phuture@acidlab:~$ git clone https://github.com/midilab/aciduino.git
phuture@acidlab:~$ cd aciduino/
phuture@acidlab:~$ git submodule update --init --recursive
```

Interplain, 2023-08-06:
Cool, I'll check it out. I will test it tomorrow.  I will get back to you
how things go. This unable to compile topics getting to be another thing
maybe it should have a kind of forum of its own. Its moving into
development.  And maybe you could use a FAQ section. I have not used
github in a while. And not sure you can arrange that. I have Discord to
would be nice to talk with you.

midilab, 2023-08-06:
`And maybe you could use a FAQ section. I have not used github in a while. And not sure you can arrange that.`
Yes totally needed, i've being running out of time to handle all those stuffs lately, but its on the roadmap to get better documentation, faq sections and so on, anyway this is a open community project everyone is invited to contribute with whatever they can.

`I have Discord to would be nice to talk with you.`
For a while i have being thinking of a forum for more open discuss, i think this is he time, i've just created a discord and maybe we can go fromthere, just search for midilab user over there.

`This unable to compile topics getting to be another thing maybe it should have a kind of forum of its own. Its moving into
development. `
Sure thing, i've always try to stay on topic here on github, when things like that comes in normally i open another issue to handle or just put it on roadmap.

Interplain, 2023-08-06:
Do you have a Discord invitation link?

midilab, 2023-08-06:
https://discord.gg/TwPR9VFZ

closed 2023-08-09 by midilab