midilab/aciduino #41: USB-MIDI settings

hugoescalpelo, 2023-11-07:
Which are the baud, data bits, parity and stop bits of USB-MIDI in Aciduino V2?

I built an Aciduino V2 for protoboard with an ESP32 DevKit, the only change I've made to the code was set the GPIO34 because my model does not have GPIO0 external. I am trying connect it to Ableton Live in Windows. For this I use [hairless-midi-serial](https://projectgus.github.io/hairless-midiserial/) and loopMIDI. 

It kinda sounds at baudrate:19200, databits:8, parity:none, stop bits:1, flow contorl:none and send this error: +7.473 - Warning: got a status byte when we were expecting 1 more data bytes, sending possibly incomplete MIDI message 0x82.

I tried default settings, clearing all notes and then leaving just a simple pattern of 8 notes in 303, and playing the same pattern as the videos, but not good results.

I tried to read all code and documentation from V1 to usb-midi definitions for uCtrl but didnt understand enough.

Please, help.

copych, 2023-11-08:
@hugoescalpelo default settings for Hairless are 115200, no flow control. It's in the "Getting started" section of the page you supplied.

midilab, 2023-11-08:
> Which are the baud, data bits, parity and stop bits of USB-MIDI in Aciduino V2?

For ESP32 family devices, there is native USB/MIDI support available for those that have TynUSB out-of-the-box support. However, Looks like your ESP32 does not have native support, you'll need to use a Hairless bridge, which it seems you are currently attempting to do.

The Aciduino V2 initializes the MIDI interface to be used as a real MIDI port, based on the Arduino MIDI library, in the file 1_uctrl.ino, line 28 on last git commit:

`...
#else
    MIDI_CREATE_INSTANCE(HardwareSerial, Serial1, MIDI1);
...`

So in this case you need to make sure the usb port of your ESP32 is Serial1, otherwise just change to the correct one.

After confirming the correct Serial port, you'll need to change the baud rate, which is automatically set by the MIDI library to 31250 (MIDI Standard definition), to match the one you intend to use on your computer. If you are working with Hairless, you can set it to 115200.

To do this, add the following line to the very end of your setup() function in the AciduinoV2.ino file:

`Serial1.begin(115200);`

Make sure Serial1 is the one on usb port side of your ESP32, otherwise change it accordly, and make sure that you're adding this line to the very end of setup() call on file AciduinoV2.ino.

Hope it helps! let me know!

closed 2023-11-22 by midilab

hugoescalpelo, 2023-11-23:
It worked!

Sorry for the delay, executive disfunction attacked. I just tried adding the `Serial.begin(115200);` at release commit `b7aa2ee` (sep 4) and it worked. Hairless MIDI does not send any more errors :D 

I will try to connect to Ableton Live and then with commit `daa7b40`, I think that's the one you mentioned. I'll keep you posted.

Do you want yo see my circuit? I hope you like it: [link](https://www.instagram.com/p/CzU3j41Op77/)

Thank You!

midilab, 2023-11-27:
Looks great! any songs or jams made on this one?

hugoescalpelo, 2023-12-08:
Yes! I have this demonstration!!

https://www.instagram.com/reel/C0nAs_hujp8/?utm_source=ig_web_copy_link&igshid=MzRlODBiNWFlZA==