midilab/aciduino #42: Aciduino V1 in Leonardo

estufilla, 2023-11-18:
Hi and sorry for asking something probably obvious but, I've read everything about midilab GH and still can't find a way. Is it possible to use a Leonardo for V1? And V2? Or a UNO?

midilab, 2023-11-18:
Hi @estufilla ,

Aciduino V1 is built on nano only, i have plans to port it to leonardo soon so we can have native usb midi port on aciduino V1, unfortunally leonardo cant handle aciduino v2 because of memory limitations.

We are always trying to port to different plataforms and now the V2 has support for arudino mega, all teensy arms and some esp32.

estufilla, 2023-11-18:
Hey! 
Fast and concise answer, the best ones! Sorry about writing in the other closed post, but was the most "resembling" thing I was searching for. I'm extremely new to coding and such, and I was getting truly frustrated. Been trying for a few days :D

I somehow made it "work" in the Uno one. Everything worked as MIDI-Serial but anything related with the tempo. I get some errors when uploading the code, still uploads it and "works". Something uClock related.

`C:\Users\+++\Desktop\aciduino-master\aciduino-master\v1\Aciduino\uClock.cpp: In member function 'void umodular::clock::uClockClass::handleTimerInt()':
C:\Users\+++\Desktop\aciduino-master\aciduino-master\v1\Aciduino\uClock.cpp:358:27: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
     onClock96PPQNCallback(&internal_tick);
                           ^~~~~~~~~~~~~~
C:\Users\+++\Desktop\aciduino-master\aciduino-master\v1\Aciduino\uClock.cpp:363:29: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
       onClock32PPQNCallback(&div32th_counter);
                             ^~~~~~~~~~~~~~~~
C:\Users\+++\Desktop\aciduino-master\aciduino-master\v1\Aciduino\uClock.cpp:366:29: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
       onClock16PPQNCallback(&div16th_counter);
                             ^~~~~~~~~~~~~~~~
C:\Users\+++\Desktop\aciduino-master\aciduino-master\v1\Aciduino\uClock.cpp:374:29: warning: invalid conversion from 'volatile uint32_t* {aka volatile long unsigned int*}' to 'uint32_t* {aka long unsigned int*}' [-fpermissive]
       onClock32PPQNCallback(&div32th_counter);
                             ^~~~~~~~~~~~~~~~
Sketch uses 14108 bytes (43%) of program storage space. Maximum is 32256 bytes.
Global variables use 715 bytes (34%) of dynamic memory, leaving 1333 bytes for local variables. Maximum is 2048 bytes.
`

Anyhow, thank you and I will try to learn as much as possible.

midilab, 2023-11-21:
I'd personaly runs my aciduino v2 on a arduino mega, more than enough for aciduino v2 needs, ESP32 is somehow still experimental since its a total different archtecture it demands more work to port and test. so far not extensively tested out there. But beta testers are always welcome too! In other words, a arduino mega will work out of the box, esp32 you migth need to adpat, read, change, debug or code something new depending on the board. not garantee at all! but a work in progress...

estufilla, 2023-11-22:
Nice thanks! 

So a Mega Mini and an ESP32, so I can learn and help.

Could you give me any hint on the compile errors? As said, with a hint or two so I could go on my own way and do it myself.

Either way, thank you again!

midilab, 2023-11-27:
you can get some help on midilab discord server: https://discord.gg/gtKWj7Hg
to avoid get out of topic im closing this issue.

closed 2023-11-27 by midilab