midilab/aciduino #44: 3D printed Case

hugoescalpelo, 2024-01-15:
Is it possible to join as collaborator?

I designed a 3D printed case for perforated prototyping PCB and I want to collaborate.

Also, I'm designing a dedicated PCB for Aciduino V2 and in the next weeks I'm also designing a 3D printed case for that circuit.

midilab, 2024-01-18:
hi @hugoescalpelo 

sure you can! please do it via pull request, so we can use the space to thread discussion before merge.

midilab, 2024-01-21:
You can send some ideas and photos here before do a pull or join discord to discuss about it, there are other people already doing aciduinos with plenty of ideas too, maybe a open discussion will benefit everybody.
https://discord.gg/JjDs28NS

hugoescalpelo, 2024-01-30:
Thank you! Sorry for the delay. The discord link is expired.

midilab, 2024-01-30:
> Thank you! Sorry for the delay. The discord link is expired.

no worries dude! https://discord.gg/EzbaPqwCJG

closed 2024-03-20 by midilab