midilab/aciduino #50: Steps to compile in Arduino IDE

copych, 2024-11-05:
Please, specify steps to compile Aciduino v2 in Arduino IDE.
Either I use folder from the master, or use the folder structure from the 2.0.0 release, I get the errors complaining on missing .h files.
I suppose that I do something wrong, so I better ask here for the instructions.
Also, please, can you advice if your code runs on just one core of the ESP32 MCU?

midilab, 2024-11-06:
hey @copych ,

If you follow the [download steps](https://github.com/midilab/aciduino/tree/master/v2#download) on readme you should be fine.

One thing missing on README is the current supported version of esp32 arduino core.

The last know to work is the [espressif core](https://docs.espressif.com/projects/arduino-esp32/) version 2.0.18

There is a [pull request](https://github.com/midilab/uClock/pull/46) for the core update that will be soon integrated.

Try with the mentioned version of espressif. If the problems persist, please send me the errors and some environment information:

Arduino IDE version:
ESP32 Module:
Download method used:

copych, 2024-11-06:
UPD: deleted ArduinoBLE from the `libraries` folder? now it compiles for ESP32 DevBoard.

Is there a way to compile for ESP32S3 ???
The matter is I've managed to fit the whole AcidBox synth (2 x 303 and a drumSampler + FX) into a single ESP32S3's core0, so core1 is free to run all the control routines.

midilab, 2024-11-06:
Any board that is not compiling should be investigate to create the correct port(ESP32 variations).

If DevBoard is not compiling with BLE we should guard it via macro to allows this version to run without BLE for example.

If you could share the board and compile errors(per board) we can try to manage some sort of port for specific esp32 models.

copych, 2024-11-06:
ESP32-S3 errors https://gist.github.com/copych/317927c757589ac1ff0ce62c578f4827

copych, 2024-11-06:
just in case, if you want to differentiate between ESP32 and ESP32S3 at compile time, you'd rather use something like this
```
#if defined(CONFIG_IDF_TARGET_ESP32S3)
// code for ESP32-S3 based boards
#elif defined(CONFIG_IDF_TARGET_ESP32)
// code for ESP32 based boards
#endif
```

and a fork for ESP-Arduino core versions:
```
#if (ESP_ARDUINO_VERSION_MAJOR < 3)
// 2.x.x version code
#else
// 3.x.x version code
#endif
```

midilab, 2024-11-08:
Thanks for the macros needed.

Looks like the problem with S3 is the BLE(you can remove it!) and the midi-usb support.

this usb-midi support were never tested because i dont own a S3, you need to enable midi usb support for S3 on IDE i guess before try to compile so espNativeUsbMidi will be avaliable to init usb midi stack.

I will try to integrate the macro setup asap.

midilab, 2024-11-09:
So i find the error for esp32s3:

the ports/esp32/wroom.h have a commented section that defines the usb midi object.

// in case we got USB native mode support builtin, use it! note the case of wroom, keep this setup for later other boards
//#if defined(CONFIG_TINYUSB_ENABLED)
//ESPNATIVEUSBMIDI espNativeUsbMidi;
//MIDI_CREATE_INSTANCE(ESPNATIVEUSBMIDI, espNativeUsbMidi, MIDI1);
//#endif

try to comment out it!

copych, 2024-11-14:
Hello, thanks for the feedback. I've copied wroom.h to s3.h and modified as you pointed.
Here are the errors:
https://gist.github.com/copych/b47cf8dc025e32c524e9b592141eb790

midilab, 2024-11-14:
The error is related to BLE-MIDI scope name clashes. You can remove BLE-MIDI and any reference from the code, it was a proof of concept implementation but its so lame for midi messages, maybe usable for CC messages but still, i think i will remove completly or leave it as a option macro to enable it.

copych, 2024-11-14:
I continue with the release ver.
errors: https://gist.github.com/copych/42e0f22c1ce68a4870d5f9a1dc19c75e

midilab, 2024-11-16:
hey @copych ,

I made some changes and some pull requests for newer usage of espressif arduino core esp32 that compiles now on S3, all the code is inside develop branch:

```console
you@acidlab:~$ git clone https://github.com/midilab/aciduino.git
you@acidlab:~$ cd aciduino/
you@acidlab:~$ git checkout develop
you@acidlab:~$ git submodule update --init --recursive
```  
I have organized the esp32 ports and created a s3 file that compiles fine, but never tested before, so please report back so we can try to start a new esp32 model support on aciduino.

You may need to adpt the s3.h file for pinage based on your board pinout map.

copych, 2024-11-18:
Hi @midilab 
I have tried to compile but failed again.
I have upgraded to ESP-Core v.3.0.7 and firstly it gave this 
```
name.c:5:10: fatal error: usb_names.h: No such file or directory
    5 | #include "usb_names.h"
      |          ^~~~~~~~~~~~~
compilation terminated.
exit status 1
usb_names.h: No such file or directory
```
To my mind this shouldn't have happened because of the guard ```#if defined(TEENSYDUINO) && defined(USB_MIDI)``` but it spawned even aftar I have emptied the ```name.c``` in the root of the project and even after deleting it. I've searched through the sources and found those 3 examples for Teensy which also contained ```name.c```, so I deleted them thou it was already beyond my understanding. After that it started scrolling this:
https://gist.github.com/copych/c2ab84a5726203367a3f3d390bb188e6
It seems that I should separate the project files from all the stuff supplied, cause Arduino seems to be recursively collecting all the files in the AciduinoV2 folder, whish is somehow confirmed by unusually long building time.

midilab, 2024-11-18:
What is the board name you are trying to compile for? (the name selected on arduino dropdown menu for board)

copych, 2024-11-18:
```Esp32s3 dev module```

midilab, 2024-11-18:
There is the examples folder along with uClock library inside src directory of AciduinoV2 that needs to be deleted, i will add to the README.md as reference.

I have no problems compiling here with the board you're using, did you change any board configuration?
Arduino 2.3.3 and espressif esp32 arduino core 3.0.7.

copych, 2024-11-18:
Now, with removed examples, I get this
https://gist.github.com/copych/accb357d49617aef1e09035a0fd34a43

midilab, 2024-11-18:
It looks like you are using main branch instead of develop.

Try to follow the develop checkout instructions again and make sure you are on develop branch before init the library dependencies.

copych, 2024-11-18:
I manually reconstructed locally what you have in `develop`, and still I have
```
In file included from D:\_CLOUD_\arduino\AciduinoV2\AciduinoV2.ino:39:
D:\_CLOUD_\arduino\AciduinoV2\src\ports\esp32\s3.h:13: warning: "LED_BUILTIN" redefined
 #define LED_BUILTIN       2
 
In file included from C:\Users\_USER_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17\cores\esp32/esp32-hal-gpio.h:29,
                 from C:\Users\_USER_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17\cores\esp32/esp32-hal.h:83,
                 from C:\Users\_USER_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17\cores\esp32/Arduino.h:36,
                 from D:\_CLOUD_\arduino\AciduinoV2\AciduinoV2.ino:28:
C:\Users\_USER_\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.17\variants\esp32s3/pins_arduino.h:17: note: this is the location of the previous definition
 #define LED_BUILTIN LED_BUILTIN  // allow testing #ifdef LED_BUILTIN
 
In file included from D:\_CLOUD_\arduino\AciduinoV2\src\uCtrl\module\midi\midi.hpp:33,
                 from D:\_CLOUD_\arduino\AciduinoV2\src\uCtrl\uCtrl.h:38,
                 from D:\_CLOUD_\arduino\AciduinoV2\src\aciduino.hpp:15,
                 from D:\_CLOUD_\arduino\AciduinoV2\AciduinoV2.ino:33:
D:\_CLOUD_\arduino\AciduinoV2\src\uCtrl\module\midi\BLE-MIDI\hardware\BLEMIDI_ESP32.h: In member function 'virtual void bleMidi::MyCharacteristicCallbacks::onWrite(BLECharacteristic*)':
D:\_CLOUD_\arduino\AciduinoV2\src\uCtrl\module\midi\BLE-MIDI\hardware\BLEMIDI_ESP32.h:116:50: error: conversion from 'std::__cxx11::string' {aka 'std::__cxx11::basic_string<char>'} to non-scalar type 'String' requested
         String rxValue = characteristic->getValue();
                          ~~~~~~~~~~~~~~~~~~~~~~~~^~
In file included from D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\acid_sequencer.h:36,
                 from D:\_CLOUD_\arduino\AciduinoV2\src\aciduino.hpp:20,
                 from D:\_CLOUD_\arduino\AciduinoV2\AciduinoV2.ino:33:
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h: In member function 'virtual uint16_t Engine::getTrackLength(uint8_t)':
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h:87:53: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint16_t getTrackLength(uint8_t track) {}
                                                     ^
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h: In member function 'virtual bool Engine::stepOn(uint8_t, uint8_t)':
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h:91:55: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual bool stepOn(uint8_t track, uint8_t step) {}
                                                       ^
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h: In member function 'virtual uint8_t Engine::getStepData(uint8_t, uint8_t)':
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h:95:63: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint8_t getStepData(uint8_t track, uint8_t step) {}
                                                               ^
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h: In member function 'virtual uint8_t Engine::getCurrentStep(uint8_t)':
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h:97:52: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint8_t getCurrentStep(uint8_t track) {}
                                                    ^
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h: In member function 'virtual uint8_t Engine::getTrackVoice(uint8_t)':
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h:101:51: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual uint8_t getTrackVoice(uint8_t track) {}
                                                   ^
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h: In member function 'virtual const char* Engine::getTrackVoiceName(uint8_t, uint8_t)':
D:\_CLOUD_\arduino\AciduinoV2\src\sequencer\engine.h:102:83: warning: no return statement in function returning non-void [-Wreturn-type]
     virtual const char * getTrackVoiceName(uint8_t track = 0, uint8_t voice = 0) {}
                                                                                   ^
In file included from D:\_CLOUD_\arduino\AciduinoV2\AciduinoV2.ino:33:
D:\_CLOUD_\arduino\AciduinoV2\src\aciduino.hpp: In member function 'uint8_t Aciduino::setSelectedTrack(uint8_t)':
D:\_CLOUD_\arduino\AciduinoV2\src\aciduino.hpp:111:88: warning: no return statement in function returning non-void [-Wreturn-type]
   uint8_t setSelectedTrack(uint8_t selected_track) { _selected_track = selected_track; };
                                                                                        ^
exit status 1
Error compiling for board ESP32S3 Dev Module.

```

copych, 2024-11-18:
oops, on this PC I have 2.0.17, I will upgrade it

copych, 2024-11-18:
here it is
https://gist.github.com/copych/b3faf977a17688130f18cb84da165bcd

closed 2024-11-21 by copych