midilab/aciduino #51: AciduinoBox

copych, 2024-11-21:
@midilab @rdridder 
My suggestion is to make a new repository (I don't know if it can both be private and grant access for you). For the time being I have a snapshot of the minimal code to test that won't compile for some reason. With this repo we should at least reach the 'proof of concept' state, so that main questions would have answers, like "If it is compiled in the end, will it fit into the ESP32S3?" or "Are there some runtime conflicts" or "How to organize inter-module communications"...

rdridder, 2024-11-21:
@copych You can also make it a public repository, does not really matter. We can work with pull requests to submit changes.

copych, 2024-11-21:
@midilab please, advise, uCtrl/module/ contains whole huge libraries like U8g2 and SdFat. What is the reason for you to keep all these files? Are they modified versions or are you afraid that the newer versions would lack some of deprecated functions?

midilab, 2024-11-21:
> @midilab please, advise, uCtrl/module/ contains whole huge libraries like U8g2 and SdFat. What is the reason for you to keep all these files? Are they modified versions or are you afraid that the newer versions would lack some of deprecated functions?

To keep maintenance simple and development less prone to errors with different versions.

> @copych You can also make it a public repository, does not really matter. We can work with pull requests to submit changes.

My idea was to start a new branch here, by looking at the acidbox source code the easiest way to integrate it, is inside aciduino project, on src/ directory.

So the code inside src/ will be a submodule pointing to original acidbox repository. Just as uClock and uCtrl...

@copych you told me about a version where you use only 1 core to run all acidbox rigth? we probrably need it to run with aciduino smoothly.

After we have this 1 core version in hands, asap, i can try to glue everything to compile and commit it to the new branch so we can start work on it.

copych, 2024-11-22:
Here's the version where I tried to integrate projects. https://github.com/copych/AciduinoBox/blob/main/AciduinoBox/
 ATTENTION! /src/ isn't full! I started uploading, but then saw how many files it has, so it's better to link them.
Anyway, @midilab  feel free to do whatever you think is right.

midilab, 2024-12-12:
Closing due to Inactivity....

closed 2024-12-12 by midilab