midilab/aciduino #53: System freezes when using external MIDI clock on ESP32

YggdrasilUrthr, 2025-05-27:
After changing the system clock to either midi1 or midi2 (implemented via a suitable midi I/O circuit) the system becomes unresponsive. The selected option keeps blinking but I cannot use any button. I had the same issue with both the latest release version and the one cloned from the repository. I also tried the develop branch but I was unable to compile the code using Arduino IDE 2.3.3 (also tested 2.3.6) and any ESP32 core from version 2.0.17 up to the latest one. 
The only changes I made to the ports file have been assigning a pin to the transport button, reversing the pot direction and switching the hw midi port from Serial2 to Serial1.

Am I doing something wrong?

midilab, 2025-05-27:
Hi @YggdrasilUrthr ,

Thanks for the report,

Can you please try with no changes on ports file and report it back? keep the original one for this first test.

YggdrasilUrthr, 2025-05-27:
Hi, thanks for your quick answer. 
I uploaded the code using the default wroom.h file included in the release version, but the issue is still there. Specifically, I can select freely midi1 and use the all the navigation button correctly but, as soon as I press play in my DAW, the system freezes as before. On the other hand, if I select midi2 as clock source the system immediatley crashes (but I think this is due to the now miswired hardware MIDI port, which is still attached to the Serial1 pins leaving the Serial2 pins unconnected). 

Since it may be relevant, I am using a NodeMCU ESP32-S2 (which mounts the supported wroom 32 chip). I originally redefined the pins in the ports file since the default hw MIDI port (Serial2) is located on the same pin as the shift button (pin 17, as defined in the repo). I can also safely rule out any fault in the external MIDI circuit since MIDI out correctly works and I also implemented a MIDI through connection which correctly loops back any MIDI message sent from the DAW.

midilab, 2025-05-27:
I'm aware that ESP32s have some issues with specific pins. Depending on the model, a pinout that works on one ESP32 can cause another to freeze — usually because some pins are used for internal functions.

Unfortunately, the USB connector on my ESP32 is broken, so I can't verify this directly. But we can try to debug it using your board. It might take a bit of time, as I'll need to analyze things on my end and then have you test the changes.

Could you please share your modified wroom.h via a Gist?

Also, here are some notes on which pins to avoid, as they can lead to unpredictable behavior:
🔗 https://randomnerdtutorials.com/esp32-pinout-reference-gpios/

midilab, 2025-05-31:
Great, that works!

The changes are already in the develop branch and are pending merge along with a few other upcoming features.

Thanks again for your help with testing, debugging, and providing the fix.

Could you please share the exact model of your ESP32? MIDI ports used and pinout? the idea is to add as many ports to extend compatibility of different ESP32 models. We'd like to list it as a known supported device in the documentation.

YggdrasilUrthr, 2025-05-31:
Great!!

I am using a NodeMCU ESP32-S v1.1 (https://www.espboards.dev/esp32/nodemcu-32s/). On the hardware side I used the same pins defined in wroom.h, keeping in my mind that in my board they are located slightly differently (pinout in last link). I then added a transport button wired to pin 13 and reversed the pot direction (I accidentally soldered it upside down, oops). I kept the hardware midi port on Serial2 (pins 16 and 17 on my board) as in the original file. I uploaded the final port configuration here: https://gist.github.com/YggdrasilUrthr/3ff72e113739ec97654d7f4f59dfdf8d 

Finally, I used this circuit https://diyelectromusic.com/2023/12/12/raspberry-pi-zero-midi-pcb/ for the midi hw interface (I just copied the circuit on breadboard without a dedicated PCB), connected to pin 16 and 17 (Serial2). I only swapped the 74HCT14 with a SN74HC14N (which according to the guide should not work, but I effectively used it without any problem on three different 3.3 devices). 

That should be everything! Let me know if you need any additional detail.